stm32duino/VL53L3CX #5: Is there any demo with peoplecouter used by VL53L3CX

iot-lorawan, 2021-01-08:
Hi,
  Is there any demo with peoplecouter used by VL53L3CX

cparata, 2021-01-08:
Hi @iot-lorawan ,
not yet. Do you know a reference library for that?
Best Regards,
Carlo

cparata, 2021-01-08:
Hi @iot-lorawan ,
I got more information from ST Imaging experts. Unfortunately, the people detection demo that you can find [here](https://my.st.com/content/my_st_com/en/products/embedded-software/proximity-sensors-software/stsw-img010.license=1610110800128.product=STSW-IMG010.version=2.0.0.html) is currently available only for VL51L1X component. So we can implement an Arduino sketch that implements the people detection demo only for the [VL53L1X](https://github.com/stm32duino/VL53L1X) component and the relative expansion board ([X-NUCLEO-VL53L1A1](https://github.com/stm32duino/X-NUCLEO-53L1A1)). Instead, it is not applicable at the moment for VL53L3CX and VL53L1 components. So I have to close this ticket because it is not applicable. If you are interested to have the people detection example for VL53L1X, please, open an issue for this library or the X-NUCLEO-VL53L1A1 library.
Best Regards,
Carlo

closed 2021-01-08 by cparata

cparata, 2021-01-08:
Hi @iot-lorawan ,
just for your information, I updated the VL53L1X and the X-NUCLEO-VL53L1A1 adding the people counting demo. If you download the last version of these 2 libraries you can get it.
Best Regards,
Carlo

iot-lorawan, 2021-01-11:
Hi，
    Thank you very much. the link https://www.st.com/en/embedded-software/stsw-img010.html is very useful for me.

Best Regards