stm32duino/VL53L3CX #7: Using ESP32 with the VL53L3CX?

mottibz, 2021-05-02:
Hi, Can anybody help in adapting this library to the ESP32? I have tried with no success. Thanks!

fpistm, 2021-05-03:
Hi @mottibz 
This library used standard Arduino API (mainly `Wire`) You should be able to use it on ESP32 without any modification.
Could you be more precise on your issue with ESP32 ? Do not build ? Could not execute properly ? ....

Anyway I saw that library.properties only give stm32 architecture support:
https://github.com/stm32duino/VL53L3CX/blob/0d083fda40b0b857c83b30b09140dcdffcb84261/library.properties#L9

@cparata is this library compatible with other architecture or is there any known restrictions ?

mottibz, 2021-05-03:
Hi, thanks for your reply. The library doc provides instructions how to connect to an Arduino (I believe). For example, the polling example code includes initializing for pin A1 ("VL53LX sensor_vl53lx_sat(&DEV_I2C, A1);"). I tried changing that to the pin number on the ESP32 that I connected it to, it compiled OK, but I got nothing happening. Any help in adapting to the ESP32 will be very highly appreciated!

fpistm, 2021-05-03:
Well honestly I don't see any adaptation to perform. As stated it uses standard Arduino API so if ESP32 is fully Arduino compatible I don't see how to help. Are you sure you properly wired your sensor ?

mottibz, 2021-05-03:
Yes, all wired fine. I am a well experienced ESP32 experimenter :-)

cparata, 2021-05-03:
Hi @mottibz ,
could you try to call "VL53LX_ClearInterruptAndStartMeasurement" instead of "VL53LX_StartMeasurement" in the sample sketch?
I would not that it is the same issue experienced with [VL53L1](https://github.com/stm32duino/VL53L1/issues/18) on an ESP platform, even if there it was the ESP8266 instead of the ESP32.
Best Regards,
Carlo

mottibz, 2021-05-03:
OK, now I am getting something :-)  Thanks!!!
I am getting lines printed like the below:
VL53LX Satellite: Count=118, #Objs=2 status=0, D=18mm, Signal=0.92 Mcps, Ambient=0.23 Mcps
                               status=0, D=1582mm, Signal=0.53 Mcps, Ambient=0.23 Mcps
I understand from that printout that it recognized two items with distance of 18mm and 1582mm. In reality, there is nothing with a distance of 18mm and the ceiling above it is with a distance of 162cm (4 cm off). 
- Is there a need to calibrate the sensor?
- I am also noticing that it reacts to distance changes fairly slow. Is there a way to get it to recognize distance changes faster?
Thanks a lot!!!

mottibz, 2021-05-03:
Hi, thanks @cparata and @dhebbeker for your help!
Any suggestions regarding my two questions above?

fpistm, 2021-05-03:
@cparata if I well understood some update are needed in the library like the ones done in VL53L1 ?

dhebbeker, 2021-05-03:
@mottibz, for the latter question: I also experienced significant delay between the change in distance and the corresponding output on the serial terminal (although with the (library for) the VL53L1). I suspect that the reason for the delay is the serial data receiver on the host PC. My assumption is based on the fact that when disconnecting the MCU, new lines were printed on the serial terminal for many more seconds. I have been using the serial console integrated in Arduino IDE. But the cause for this may root in lower levels of the software / operating system.

I suggest to reduce the frequency of the serial output. This should allow the receiver to empty its buffer. Here is how I did it with the VL53L1: dhebbeker/VL53L1@fa2992437af5387d74674dec7d9bfbc24875ef70 Can you confirm my findings?

mottibz, 2021-05-03:
Hi David, first, I am using an ESP32 which is higher performance than the ESP8266. Second, I am using a powerful PC. Third, I added something else to reduce the number of readings and printing to the serial port. I just added a "delay(1000)" as the last line of the loop() function (I am using the polling sample, the interrupt version should not make a change and it also requires some changes to make it thread safe). Bottom line is that I am still seeing a long delay of 3-4 seconds from the time of actual change until it is reported which can't be related (in my opinion) to the serial printing. It can be either connected to how the I2C bus is being utilized by the sensor library or how the sensor accesses the I2C bus from its side, or to the way that the sensor performs the reads. Which such delays, it makes the sensor useless. We need some help from the experts...

mottibz, 2021-05-04:
I found a few library functions that provide some info about the chip. Below is printout of the info I got. One strange thing is that in the library structure definition for GetDeviceInfo(), the Module Type is listed in a comment as should be 0xAA but I am getting 0. Anyone familiar with this and can comment (as I am still getting the slow recognition which makes the sensor useless as it behaves now)? Printout:

GetVersion: rc=0, Build=4, Major=1, Minor=1, Revision=2352
GetProductRevision: rc=0, Major=1, Minor=0
GetDeviceInfo: rc=0, ModuleType(0xAA?)=0x0, ProdRevMajor=1, ProdRevMinor=0
GetUID: rc=0, UID=0x6F7250202C313D

Thanks for any help!

cparata, 2021-05-04:
Hi all,
unfortunately, I do not have any ESP board to make some tests on my side. Actually, with Nucleo boards I did not find any particular delay to get the output of the sensor and in general the performances seem to me in line with the X-Cube code available on st.com for the specific component. I can ask to ST VL53L3CX experts to get a feedback on that.
Best Regards,
Carlo

mottibz, 2021-05-04:
Hi Carlo, Your help in resolving this will be very highly appreciated!
If you want, I can test for you whatever is being suggested.
BTW, I tried to set the I2C bus to 400K instead of the default 100K ("Wire.setClock(400000);") but it did not change anything.
Thanks a lot!

fpistm, 2021-05-04:
Just to be inline adding a delay of 1000 in your loop mean you do a read each 1 second. Right?
I think the point raised by @dhebbeker  is correct as without any delay you made read consecutively and I guess Serial is not able to flush all at real time.
You could try to store several reading in a buffer then flush it.

mottibz, 2021-05-04:
Hi, I am moving to using the interrupt method and research it. Will report out ASAP

mottibz, 2021-05-04:
https://github.com/mottibz/ESP32-VL53L3CX-Test-app

mottibz, 2021-05-04:
Hi all, I found out that the polling example, following modification to support the ESP32 has major measurement delays so I moved to an interrupt driven approach which shows much better results.
The provided link is a test app that I developed (taking parts of the examples provided by the library), supporting the ESP32 with interrupts. It also has a recovery mechanism for lost interrupts. Please try and let me know if that works for you.
https://github.com/mottibz/ESP32-VL53L3CX-Test-app

fpistm, 2021-06-22:
AS it seems you are able to use the lib, I close this issue.

closed 2021-06-22 by fpistm

yahyatawil, 2021-12-26:
For reference, I could not make the library works reliably with ESP32 without increasing the I2C buffer size from Wire.h as mentiond in [DFRobot wiki](https://wiki.dfrobot.com/DFRobot_VL53L3CX_ToF_Distance_Sensor_Breakout_Board_SKU_SEN0378).

cparata, 2022-01-03:
Hi @yahyatawil ,
could please check if the library that you can find [here](https://github.com/cparata/VL53L3CX) fixes the issue on ESP32 without any change on Wire library? If so, I will push the modifications also on the main branch of the STM32duino library and I will generate a new release. Please, let me know.
Best Regards,
Carlo

yahyatawil, 2022-01-05:
Yes. I will try the mentioned branch and report back shortly.

yahyatawil, 2022-01-16:
@cparata The fork did not work. I tried the VL53L3CX_Sat_HelloWorld example with and without the Wire buffer change.

cparata, 2022-01-17:
Hi @yahyatawil ,
it's very strange. I tested the new driver with a Nucleo board and the X-NUCLEO-53L3A2 and it works without any issue.
The driver now should be able to use a Wire library with whatever BUFFER_LENGTH. What is the default I2C_BUFFER_LENGTH on ESP32? The only explanation is that it is less than 32 bytes. If it is the case you can try to edit the vl53lx_class.cpp and replace the following defines:

```
#ifndef DEFAULT_I2C_BUFFER_LEN
  #ifdef BUFFER_LENGTH
    #define DEFAULT_I2C_BUFFER_LEN  BUFFER_LENGTH
  #else
    #define DEFAULT_I2C_BUFFER_LEN  32
  #endif
#endif
```

putting the DEFAULT_I2C_BUFFER_LEN to the default value foreseen on ESP32.
Best Regards,
Carlo

yahyatawil, 2022-01-17:
The `I2C_BUFFER_LENGTH` is 128, I incease it to 512 with orginal stm32duino branch and did work. For the fork did not work either with the increase or with the 128 default value. Anyway, If I had free time soon, I will try it again.

cparata, 2022-01-17:
Ok, so in my opinion it means that ESP32 Wire library does not work well when the I2C transaction is split in many small transactions. I guess that the issue with the new code is due to the fact that if in the ESP32 Wire library the foreseen defines are not defined, the default buffer length will be 32 bytes that is too small for the ESP32 Wire library