stm32duino/VL53L3CX #11: stability issue with esp32-cam.

Pratiksha0102, 2022-05-16:
Hello!

I am getting trouble While getting range from the sensor using esp32-cam. It works fine for while but after sometime it gives status error 255 and 12 and the problem is that the device is placed at one place and detecting same object but after sometimes it stops detecting the same object and give status code 255 and 12. I don't understand the behavior of the sensor. Is there any hardware issue or library issue?  I am using esp32-cam and vl53l3cx Time of flight sensor.

[image]
[image]

cparata, 2022-05-16:
Hello @Pratiksha0102 ,
it's difficult to debug because it depends on your hardware setup. When you get a range status different from 0, it means that the object is not clearly seen by the sensor and the range measurement could be not reliable. Maybe it could be an hardware issue on the sensor, if it works for a while and then it stops to work properly.
I have never seen this issue on my side, but I have never tested the library with ESP32 platforms. I always used STM32 boards.
Best Regards,
Carlo

Pratiksha0102, 2022-05-17:
Hello! @cparata,  
Thanks for quick response. We moved with vl53l1cx and it works quite well with esp32 platform.
Thanks & Regards,
Pratiksha.

closed 2022-05-17 by Pratiksha0102