stm32duino/VL53L3CX #12: Get Raw Histogram Data

JGarrett7, 2023-06-29:
For our application, we need to get the raw histogram data from our TOF sensor. I assume that the "VL53LX_GetAdditionalData" function is what does this, but it seems to have been commented out in "vl53lx_class.h". I tried uncommenting it, but am getting errors. Am I missing something else that needs to be uncommented or is this function not working for the Arduino implementation?

Thanks for any help you can give.

For those curious, the reason is the same as in this publication "Multitouch Pressure Sensing With Soft
Optical Time-of-Flight Sensors": https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=9674750

cparata, 2023-07-11:
Hello @JGarrett7 ,
Unfortunately, this feature is not available for the VL53L3CX ToF sensor. The new ToF components VL53L7CH and VL53L8CH should support this feature but unfortunately we do not still have the Arduino libraries for these sensors .  
Best Regards,
Carlo

closed 2023-07-11 by cparata

JGarrett7, 2023-07-13:
If you could look at the paper I linked, I think you'll see that it is possible. Also, when using the STM32 Cube IDE, the function exists and the documentation indicates you get histogram data from it. I can switch to using that IDE, but I thought it would be nice to use Arduino since developing with it is faster.

cparata, 2023-07-14:
Hello @JGarrett7 ,
unfortunately, I don't have the access to this article. Do You mean that the authors of this article were able to get histogram data using the Cube framework and the "VL53LX_GetAdditionalData" function?
I'm not expert of this component but the information that I provided above was given by ST ToF experts.
Best Regards,
Carlo

JGarrett7, 2023-07-14:
Sorry about that. I guess our institution gave my computer access without my realizing the article wasn't open access. This should be an open access link: https://ieeexplore.ieee.org/iel7/19/9717300/09674750.pdf?casa_token=E68gBfECYNgAAAAA:DwKt-co5anClUS8YDRiHNjggvnazCrn659bB3RSEqAS1Q3GUAjdZVxFgmZeLDHLZqpcMxQpTSng

From what I can tell though, yes. I'm not sure if they used the Arduino library or the STM32 Cube library. I will look more into it from the straight c-coding side and let you know if I manage to access the data. Thanks for your help.

JGarrett7, 2023-08-17:
So, I think I figured it out. In the paper I sent, the researchers don't use this repository or the STM32CubeIDE. They use the VL53L3CX GUI. They turn it on and then connect to the port using Putty or some other terminal. That port then displays histogram data. I'm not sure what code the GUI flashes to the Nucleo, but whatever it is can access the VL53L3CX's histogram data. If anyone finds what code that GUI is using, I'll be looking for that. Here's the terminal results 
[image]

cparata, 2023-08-18:
Hello @JGarrett7 ,
thanks for the update. Unfortunately, I do not have access to the VL53L3CX GUI code, so I don't know which firmware is flashed into the Nucleo board by the tool.
Best Regards,
Carlo