stm32duino/VL53L3CX #14: Multiple sensors example

Dario-Ciceri, 2023-12-14:
Hi, I am trying to manage two sensors, if I do an i2c scan I see a sensor at 0x29 (the other sensor I keep off by putting xshut to HIGH) but the moment I run vl53lx.InitSensor(0x30); I get 0x18 with an i2c scan instead of 0x30 and I don't understand why, I don't know if I am doing something wrong but if you have ideas you would do me a favor, thanks!

cparata, 2023-12-14:
Hello @Dario-Ciceri ,
it's normal. The address that you assign in the "InitSensor" is in 8-bits format. The value that you read in the I2C scanner is in 7-bits format. 0x18 is 0x30 shifted by 1. So everything should be fine. Consider when you assign the I2C address in the "InitSensor" that you have actually 2 addresses allocated (one for the write operation and one increased by one for the read operation ); for example, when you call `vl53lx.InitSensor(0x30);` you are allocating the 0x30 address for the write operations and the 0x31 address for the read operations. So, if you want to assign a new I2C address to another ToF sensor you must pass as parameter at least 0x32.
I hope that it can help you.
Best Regards,
Carlo

Dario-Ciceri, 2023-12-14:
> Hello @Dario-Ciceri , it's normal. The address that you assign in the "InitSensor" is in 8-bits format. The value that you read in the I2C scanner is in 7-bits format. 0x18 is 0x30 shifted by 1. So [...]

Thanks Carlo, I hadn't thought about the 1bit shift actually and I was also assigning 0x30 and 0x31 as addresses... I can try 0x30 and 0x3F to be sure, I'll try now, in the meantime thanks.

Dario-Ciceri, 2023-12-14:
If I wanted to test only one sensor at a time, without being able to disconnect them from the board, I would just turn off one sensor via xshut right?

cparata, 2023-12-14:
Hello @Dario-Ciceri ,
yes you can have all sensors connected. You just need to put low all the xshut pins and then initializing the sensors one by one.
Best Regards,
Carlo

Dario-Ciceri, 2023-12-14:
> Hello @Dario-Ciceri , yes you can have all sensors connected. You just need to put low all the xshut pins and then initializing the sensors one by one. Best Regards, Carlo

Thank you

Dario-Ciceri, 2023-12-14:
I set to low on both XSHUT sensors and then called sxTof = new Vl53lxWrapper(sxTofXshutPin,
                            sxTofInterruptPin,
                            sxTofAddress,
                            SDA_PIN,
                            SCL_PIN);

  dxTof = new Vl53lxWrapper(dxTofXshutPin,
                            dxTofInterruptPin,
                            dxTofAddress,
                            SDA_PIN,
                            SCL_PIN);

however it doesn't seem to be working, if it's not too much to ask it would be helpful to have an example with two sensors, in the meantime I'll try to see if I can figure it out

Dario-Ciceri, 2023-12-14:
> Hello @Dario-Ciceri , You can give a look at the example [here](https://github.com/stm32duino/X-NUCLEO-53L3A2/blob/main/examples/X_NUCLEO_53L3A2_HelloWorld/X_NUCLEO_53L3A2_HelloWorld.ino). It is based on a shield that mounts 3 VL53L3CX sensors with an IO expander. Anyway, I think that you can check how the several sensors are initialized to get them working. Best Regards, Carlo

I checked and I see that VL53LX_Off() is used which from what I see sets gpio0 to low, I thought it was enough to set XSHUT to low and that gpio0 was the interrupt pin (input pin)

cparata, 2023-12-14:
Hello @Dario-Ciceri ,
gpio0 pin in the VL53L3CX library should be the xshut pin. The interrupt pin is not managed inside this library but only at application level.
Best Regards,
Carlo

Dario-Ciceri, 2023-12-14:
> Hello @Dario-Ciceri , gpio0 pin in the VL53L3CX library should be the xshut pin. The interrupt pin is not managed inside this library but only at application level. Best Regards, Carlo

yes my fault!

Dario-Ciceri, 2023-12-14:
you want to laugh? after 5 hours of programming in the cold, moreover I have the flu, I realized a delay that obviously with interrupts doesn't go very well and was the cause of all evil... thanks again... I'd say it's better if I take a break for today after certain mistakes...

closed 2023-12-14 by Dario-Ciceri

martincerven, 2025-02-11:
> Hello [@Dario-Ciceri](https://github.com/Dario-Ciceri) , You can give a look at the example [...]

Thanks for this example @cparata , super easy, and worked on the first try!

jakabo27, 2025-02-16:
Hey everyone, here's a working simple example for using two sensors together to just get the distance.  When using the[ DFROBOT breakout board](https://www.dfrobot.com/product-2256.html) it uses ~80uA per sensor in sleep mode, and 16mA while taking a measurement. 

For anyone trying to use these in a battery powered application, I also set the distance mode to SHORT and the timing budget to 5000us (5ms) which reduced the awake time by nearly half.  So now it takes about 50ms for a sensor to wake up and get distance.  
 
```c
#include <Arduino.h>
#include <Wire.h>
#include <vl53lx_class.h>

#define DEV_I2C Wire
#define SerialPort Serial

const int XSHUT_PIN_A = 18;
const int XSHUT_PIN_B = 9;
VL53LX DistanceSensor_A(&DEV_I2C, XSHUT_PIN_A);
VL53LX DistanceSensor_B(&DEV_I2C, XSHUT_PIN_B);

void setup()
  {
  // Initialize serial for output.
  SerialPort.begin(115200);
  SerialPort.println("Starting!");

  DEV_I2C.begin();  // Initialize I2C bus.
  DEV_I2C.begin(SDA, SCL, 50000);  // Initialize I2C bus.

  DistanceSensor_A.begin();            // Configure VL53LX satellite component.
  DistanceSensor_A.VL53LX_Off();       // Power off VL53LX satellite component. (XSHUT LOW)
  DistanceSensor_A.InitSensor(0x23);   //Initialize VL53LX satellite component.
  DistanceSensor_A.VL53LX_SetDistanceMode(VL53LX_DISTANCEMODE_SHORT);
  DistanceSensor_A.VL53LX_SetMeasurementTimingBudgetMicroSeconds(5000); // ~500mm max distance to reduce awake time
  // https://github.com/stm32duino/VL53L3CX/blob/a3c2652e7df91b3cbcda214180910d7f143dd020/src/vl53lx_class.h

  DistanceSensor_B.begin();            // Configure VL53LX satellite component.  Sets XSHUT as output
  DistanceSensor_B.VL53LX_Off();       // Power off VL53LX satellite component. (XSHUT LOW)
  DistanceSensor_B.InitSensor(0x12);   //Initialize VL53LX satellite component.
  DistanceSensor_B.VL53LX_SetDistanceMode(VL53LX_DISTANCEMODE_SHORT);
  DistanceSensor_B.VL53LX_SetMeasurementTimingBudgetMicroSeconds(5000); // ~500mm max distance to reduce awake time

}

void loop()
{
  unsigned long startTime = millis();

  // Sensor A
  DistanceSensor_A.VL53LX_StartMeasurement();
  DistanceSensor_A.VL53LX_StartMeasurement();
  VL53LX_MultiRangingData_t MultiRangingData1;
  VL53LX_MultiRangingData_t *pMultiRangingData1 = &MultiRangingData1;
  uint8_t NewDataReady1 = 0;
  int status1;
  do{ status1 = DistanceSensor_A.VL53LX_GetMeasurementDataReady(&NewDataReady1);} while(!NewDataReady1);
  DistanceSensor_A.VL53LX_StopMeasurement();
  if((!status1)&&(NewDataReady1!=0))
  {
    status1 = DistanceSensor_A.VL53LX_GetMultiRangingData(pMultiRangingData1);
    SerialPort.print(pMultiRangingData1->RangeData[0].RangeMilliMeter);SerialPort.print("mm");
  }

  // Sensor B
  DistanceSensor_B.VL53LX_StartMeasurement();
  DistanceSensor_B.VL53LX_StartMeasurement();
  VL53LX_MultiRangingData_t MultiRangingData;
  VL53LX_MultiRangingData_t *pMultiRangingData = &MultiRangingData;
  uint8_t NewDataReady = 0;
  int status;
  do{ status = DistanceSensor_B.VL53LX_GetMeasurementDataReady(&NewDataReady);} while(!NewDataReady);
  DistanceSensor_B.VL53LX_StopMeasurement();

  if((!status)&&(NewDataReady!=0))
  {
    status = DistanceSensor_B.VL53LX_GetMultiRangingData(pMultiRangingData);
    SerialPort.print("\t");
    SerialPort.print(pMultiRangingData->RangeData[0].RangeMilliMeter);SerialPort.print("mm\t");
  }

  unsigned long endTime = millis(); // takes about 52ms awake
//  SerialPort.print("\t Time awake: "); 
  SerialPort.print(endTime - startTime); SerialPort.print(" ms");
  SerialPort.println();

   delay(1500); // Uses 160uA in sleep mode, then 32mA for 104ms
}
```