stm32duino/VL53L3CX #15: VL53L3CX stops sending data to board once analogRead is used

HelterSkelter4, 2025-01-11:
Hello,

I would like to report an issue using this library and analogRead on my custom PCB using ESP32-S3. When I upload your "VL53L3CX_Sat_HelloWorld_Interrupt" code (with my custom pin configuration ) everything works like a charm (as you can see in the picture below):

[image]

The issue arises when I try to call the analogRead function in the code. The only thing that is changed from the code that works is that I have added "Serial.println(analogRead(5));" after "if (interruptCount){" line. :

`
#include <Arduino.h>
#include <Wire.h>
#include <vl53lx_class.h>
#include <string.h>
#include <stdlib.h>
#include <stdio.h>
#include <stdint.h>
#include <assert.h>
#include <stdlib.h>

#define DEV_I2C Wire
#define SerialPort Serial

#ifndef LED_BUILTIN
#define LED_BUILTIN 13
#endif
#define LedPin LED_BUILTIN

#define interruptPin 40

// Components.
VL53LX sensor_vl53lx_sat(&DEV_I2C, 15);

volatile int interruptCount=0;

void measure()
{
   interruptCount=1;
}

void setup()
{
   VL53LX_Error status;
   // Led.
   pinMode(LedPin, OUTPUT);
   pinMode(interruptPin, INPUT_PULLUP);
   attachInterrupt(interruptPin, measure, FALLING);

   // Initialize serial for output.
   SerialPort.begin(115200);
   SerialPort.println("Starting...");

   // Initialize I2C bus.
   DEV_I2C.begin(4,5);

   // Configure VL53LX satellite component.
   sensor_vl53lx_sat.begin();

   // Switch off VL53LX satellite component.
   sensor_vl53lx_sat.VL53LX_Off();

   // Initialize VL53LX satellite component.
   status = sensor_vl53lx_sat.InitSensor(0x52);
   if(status)
   {
      SerialPort.println("Init sensor_vl53lx_sat failed...");
   }

   sensor_vl53lx_sat.VL53LX_StartMeasurement();
}

void loop()
{
   VL53LX_MultiRangingData_t MultiRangingData;
   VL53LX_MultiRangingData_t *pMultiRangingData = &MultiRangingData;
   uint8_t NewDataReady = 0;
   int no_of_object_found = 0, j;
   char report[64];

   if (interruptCount)
   {
      **Serial.println(analogRead(5));**
      int status;

      interruptCount=0;
      // Led blinking.
      digitalWrite(LedPin, HIGH);

      status = sensor_vl53lx_sat.VL53LX_GetMeasurementDataReady(&NewDataReady);
      if((!status)&&(NewDataReady!=0))
      {
         status = sensor_vl53lx_sat.VL53LX_GetMultiRangingData(pMultiRangingData);
         no_of_object_found=pMultiRangingData->NumberOfObjectsFound;
         snprintf(report, sizeof(report), "Count=%d, #Objs=%1d ", pMultiRangingData->StreamCount, no_of_object_found);
         SerialPort.print(report);
         for(j=0;j<no_of_object_found;j++)
         {
            if(j!=0)SerialPort.print("\r\n                   ");
            SerialPort.print("status=");
            SerialPort.print(pMultiRangingData->RangeData[j].RangeStatus);
            SerialPort.print(", D=");
            SerialPort.print(pMultiRangingData->RangeData[j].RangeMilliMeter);
            SerialPort.print("mm");
            SerialPort.print(", Signal=");
            SerialPort.print((float)pMultiRangingData->RangeData[j].SignalRateRtnMegaCps/65536.0);
            SerialPort.print(" Mcps, Ambient=");
            SerialPort.print((float)pMultiRangingData->RangeData[j].AmbientRateRtnMegaCps/65536.0);
            SerialPort.print(" Mcps");
         }
         SerialPort.println("");
         if (status==0)
         {
            status = sensor_vl53lx_sat.VL53LX_ClearInterruptAndStartMeasurement();
         }
      }

      digitalWrite(LedPin, LOW);
   }
}
`

What I would expect is that when the new data is ready, the sensor interrupts the processor, the analog value from pin 5 is printed, and the device prints distances, the number of objects, etc. The code then repeats once the sensor has new data.

But what happens is that the sensor interrupts the processor once, prints analog value, prints distance, the number of objects, etc., but the sensor never interrupts the device again:

[image]

Do you maybe have any idea why this happens?

Thank you in advance,
Marin

*****Lines also changed due to custom PCB (but with these changes it works without analogRead())*****
#define interruptPin 40
VL53LX sensor_vl53lx_sat(&DEV_I2C, 15);
DEV_I2C.begin(4,5);
status = sensor_vl53lx_sat.InitSensor(0x52);

HelterSkelter4, 2025-01-11:
Sorry for only part of the code being in its box. For the life of me, I cannot figure out why this happens.

cparata, 2025-01-13:
Hello @HelterSkelter4 ,
it's seems to me that you are trying to do an analogRead on pin 5 that is also the SCL pin of your hardware setup. Maybe it can cause an hardware conflict or some issue in the I2C configuration. I don't think that it's an issue of the VL53L3CX library.
Could you try to change your analog pin or your SCL pin?
Best Regards,
Carlo

cparata, 2025-01-16:
Hello @HelterSkelter4 ,
can I close this ticket?
Best Regards,
Carlo

closed 2025-01-17 by fpistm